peterhinch/micropython-samples #2: Class Variables for SSD1306 Writer

astrilchuk, 2017-12-31:
Hi Peter,

In the SSD1306 Writer class there are class level variables for text position and clipping.  I'm uncertain why these would be class level.  What if you wanted to use 2 displays, 1 with clipping disabled and one with clipping enabled?  Also 2 displays would share the same text position.  Perhaps they should be instance level?

Thanks for creating this class!

Adrian

peterhinch, 2018-01-01:
Hi Adrian,

My anticipated use case was of multiple Writer instances sharing the same physical display. This addresses the situation where multiple fonts are used on a single display, and was the reason for the class variables and methods. The point you make about multiple displays is valid. Feel free to propose a multi display version.

Pete

astrilchuk, 2018-01-01:
Hi Peter,

I didn't have a specific use-case for multiple displays but that was my first thought against using class-level variables for those parameters (and having to set class-level variables while passing an instance around felt awkward).  Now I understand you had a valid (and understandable) reason for choosing that route.

I'll probably be using the Writer class in a number of upcoming projects so I'll revisit it and give some thought to how I could approach multi-display support with multiple fonts.

Thanks again for creating this class.  I'm currently using it to display temperature and humidity readings around my house.

Adrian

peterhinch, 2018-01-02:
Hi Adrian,

I think it could be accomplished by making the class variables members of a list, and storing that list in a class member list. All methods would need to take a display_id value defaulting to 0, with that being the index into the outer list. Whether this is worth doing is moot. The Writer class was always intended to be an example of how this functionality could be achieved rather than a complete polished class, which is why it lives in my samples repository. So I always anticipated users would extend or subclass it as required.

I'm glad you're finding it useful.

Pete

peterhinch, 2018-08-14:
@astrilchuk You may like to look at [this update](https://github.com/peterhinch/micropython-font-to-py). The `Writer` class now supports multiple displays (and other improvements).

astrilchuk, 2018-08-14:
Wow, thanks Peter.  I will definitely have a look!  I never did end up using multiple displays so I'm sorry I didn't close this ticket earlier - but it's nice to know it's a feature that's now available.

closed 2018-08-14 by astrilchuk