peterhinch/micropython-samples #3: Flip Screen of SSD1306

ppontre, 2018-02-14:
Hi Peter, 
I can't sort out how to flip screen on Micropython ESP32 using the library.
I've got a TTGO board with ESP, Battery, SSD1306.
Any suggestion ?
Thanx in advance
Paolo

peterhinch, 2018-02-15:
Hi Paolo,
it can be done but it would be a substantial rewrite. I can't see a simple way to fix **_printchar()** but a rewritten **_printchar_bitwise()** function could invert each character. The **_newline()** method would need adapting so that text flow was reversed.
Pete

ppontre, 2018-02-15:
Thanx Pteter for yr prompt answer.
Here’s my update:

  1.  There’s some trick, consisting in 2 commands: flip (vert) and mirro (horiz) the entire display.

After these commands, no other change is required.

Unfortunately i could not succeed in finding and issuing the command

  1.  It seems there’s a trade-off between beautiful fonts and flip feature: yuou can’t have both.

I succeeded in flipping a display arranged with a 8x8 font, no auto-cr, all-manual….

  1.  This evening i’ll have another trial
Regairds
Paolo

Inviato da Posta<https://go.microsoft.com/fwlink/?LinkId=550986> per Windows 10

robhamerling, 2018-08-02:
@Paolo

> I succeeded in flipping a display arranged with a 8x8 font, no auto-cr, all-manual…. 1. This evening i’ll have another trial 

Hi Paolo, would you be so kind to publish your fix?
Besides an SSD1306 I'm also using an SH1106 for which the driver has a rotate method builtin, which serves my needs. I would like a similar solution for a SSD1306 display just as your described here.
Regards, Rob.

peterhinch, 2018-08-14:
@ppontre @robhamerling You may like to look at [this update](https://github.com/peterhinch/micropython-font-to-py): the `CWriter` class supports upside down display on any device whose driver is subclassed from `frmebuf` (e.g. SSD1306).

robhamerling, 2018-08-14:
@peterhinch Thanks for this pointer, looks very nice. However currently I'm satisfied with the fixed 8x8 font and needed only flipping. Maybe useful for a new project. 
BTW:  In the meantime I found how to perform flipping by modifying a few bits in ssd1306 registers. I opened an issue (#4015) at github.com/micropython and suggested a solution.

peterhinch, 2018-08-15:
My update aims for generality by supporting any display whose driver is subclassed from `framebuf`.

Alas inverting the display means using the `CWriter` class. This is over an order of magnitude slower than `Writer` as it renders glyphs one pixel at a time rather than via bit blitting.

Your SSD1306-specific method evidently won't carry that speed penalty. It also has the advantage that the same coordinates apply to graphics and text, whereas mine only flips text.

closed 2018-08-15 by peterhinch