peterhinch/micropython-samples #4: Connect with SPI, TypeError: function expected at most 5 arguments, got 6

dukeduck1984, 2018-05-09:
Hi Peter,

I copied and pasted the files (freesans20.py, ssd1306.py, writer.py and ssd1306_test.py) per your instruction, and connected the pins per the comments (using SPI) in ssd1306_test.py, and run test(True), but I got:

TypeError: function expected at most 5 arguments, got 6. 

I tried to track it according to the line number, appearantly it came from: ssd = SSD1306_SPI(WIDTH, HEIGHT, spi, pdc, prst, pcs), so I checked class SSD1306_SPI(SSD1306) in ssd1306.py but couldn't understand why it gave an error - everything looked right.

I was using a PYB Nano.

Thanks for the help in advance.
Kaiyuan

peterhinch, 2018-05-09:
I suspect that you're running an old version of firmware: there was a change around the new year which affected the SSD1306 driver. A prior version will produce these apparently inexplicable errors. Please upgrade your firmware to the current build and try again.

dukeduck1984, 2018-05-10:
Hi Peter,

Thank you.  You were correct.  It worked after the upgrade.

Thanks for the help.
Kaiyuan