peterhinch/micropython-samples #5: Inverse Colour

diginfo, 2018-05-16:
Hi;

I have just integrated the writer class with a driver for a 1.54" epaper display, however using the printstring() results in white text on a black background.

Is it possible to use the inverse, i.e. white blackground with black text ?

peterhinch, 2018-05-17:
This is not something I've looked at and I'm not in a position to follow this up at the moment. There is some discussion around this in [the forum](https://forum.micropython.org/viewtopic.php?f=14&t=3393&start=40) and hopefully someone will investigate.

If you feel like experimenting, the Writer class has code for bitwise rendering - change **printstring** to use **_printchar_bitwise()**. On the face of it this method could easily be adapted for black on white but you'd have to first fill the framebuf with white. Bitwise rendering is slow, but epaper is slow at the best of times. So that's the approach I'd use.

diginfo, 2018-05-18:
Hi Peter;

I have managed to solve this quite easily, see the invert code below:

```
  def printstring(self, string, invert=False):
    for char in string:
      self._printchar(char,invert)

  # Method using blitting. Efficient rendering for monochrome displays.
  # Tested on SSD1306.
  def _printchar(self, char, invert):
    if char == '\n':
      self._newline()
      return
    glyph, char_height, char_width = self.font.get_ch(char)

    if Writer.text_row + char_height > self.screenheight:
      if Writer.row_clip:
        return
      self._newline()
    if Writer.text_col + char_width > self.screenwidth:
      if Writer.col_clip:
        return
      else:
        self._newline()
    buf = bytearray(glyph)
    
    if invert:
      for i, v in enumerate(buf): buf[i] = 0xFF & ~ v
    fbc = framebuf.FrameBuffer(buf, char_width, char_height, self.map)
    self.device.blit(fbc, Writer.text_col, Writer.text_row)
    Writer.text_col += char_width
```

peterhinch, 2018-05-18:
Thanks for that. I've updated the repo - the only change being to default **invert** False to avoid breaking existing code.

closed 2019-02-18 by peterhinch