peterhinch/micropython-samples #10: [ESP8266]: Error when call >>> get_time(True) <<<

hasenradball, 2019-06-30:
Hi,

when calling `get_time(True)` I get the Error "RTC Object has no attribute rtc", in file <ds3231_port.py> on line 83.

``` rtc.init((YY, MM, DD, hh, mm, ss))```

Is this a bug?

Best regards Frank

peterhinch, 2019-06-30:
I think you are running an old version. That line number doesn't correspond to current code which reads: https://github.com/peterhinch/micropython-samples/blob/71fcc81822ba150a6e15ece802a091ee0a462a31/DS3231/ds3231_port.py#L83

Have you run **ds3231_port_test.py**? You'll need to change the pin numbers for ESP8266 https://github.com/peterhinch/micropython-samples/blob/71fcc81822ba150a6e15ece802a091ee0a462a31/DS3231/ds3231_port_test.py#L23-L24

hasenradball, 2019-06-30:
Hi Peter,

thanks for the fast reply.
I downloaded the code today in the morning.
Ok the line number seems not to fit, ok. But the problem on the ESP8266 is: 

The rtc.init() is not available.
On the ESP8266 the call of:

rtc.datetime() works fine

The same as for the pyboard.

Frank

hasenradball, 2019-06-30:
Hi Peter,

I check again the line # was "78" in the method convert()

Original Code:
```
if set_rtc:
            if rtc is None:
                # Best we can do is to set local time
                secs = utime.mktime(result)
                utime.localtime(secs)
            else:
                if sys.platform == 'pyboard':
                    rtc.datetime((YY, MM, DD, wday, hh, mm, ss, 0))
                else:
                    rtc.init((YY, MM, DD, hh, mm, ss))
        return result
```
Correction for the ESP8266 with micropython 1.11:
```
if set_rtc:
            if rtc is None:
                # Best we can do is to set local time
                secs = utime.mktime(result)
                utime.localtime(secs)
            else:
                if sys.platform == 'pyboard':
                    rtc.datetime((YY, MM, DD, wday, hh, mm, ss, 0))
                else:
                    rtc.datetime((YY, MM, DD, wday, hh, mm, ss, 0))
                    #rtc.init((YY, MM, DD, hh, mm, ss))
        return result
```
 Works fine!

Best Regards

Frank

peterhinch, 2019-07-01:
Thank you for the report. I've posted an update with: https://github.com/peterhinch/micropython-samples/blob/ac7335b66b32dc4c528ca80947950db2f9b2bc62/DS3231/ds3231_port.py#L75-L78
I'd be grateful if you could try this and report back.

hasenradball, 2019-07-01:
Hi Peter,

exactly the same change I would have proposed.
I will check this.

Thanks

hasenradball, 2019-07-02:
Hi Peter it works fine!
Thank you!

peterhinch, 2019-07-03:
Great. I've re-tested against latest firmware on ESP32 so I'll close this now. Thank you for raising this and testing.

closed 2019-07-03 by peterhinch

hasenradball, 2019-07-03:
Hi Peter,

in expectation that you are an micropython expert, maybe you can answer my question regarding the Pins on the ESP regarding the IC2 bus.
I want to use the DS3231 and the DHT22 on the ESP8266-01, there I have the following Pins available:

- Pin 0
- Pin 1 (UART - TX)
- Pin 2
- Pin 3 (UART - RX)

Normally for measuring the temperature and maybe control the chip I need one Pin for the DHT22 signal, and two Pins for the I2C and if possible one Pin for the UART  - TX functionality.
Can you answer if it is possible:
- to use the Pins 1 and 3 for I2C functionality
- only to make the TX functionality Pin available on Pin 2 for example
- and use the DHT on Pin 0

If possible can you say how to set up the Pin config fot the UART?

Thank you
Best regards

Frank

peterhinch, 2019-07-03:
This would be better raised in the forum: I have no experience of re-purposing the ESP8266 UART pins, but others have. A forum search may turn something up.

Re pins 0 and 2 these have pullups which are required for correct boot. This makes them well suited for I2C but some boards have LED's on these pins which will flash when I2C is working. But as I2C drivers are open drain and start out in an **off** state the boot process works. That's as far as my own experience takes me here.

hasenradball, 2019-07-03:
Hi Peter,
so you would say Pin 0 + 2 should be preferred for I2C, right?
Is there an separate Pullup necessary when using DS3231 board (when I measure the resistance Between VCC + SCL, VCC + SDA --> I measure about 4,7 k Ohm), so I think there are pull ups available on the DS3231 board. right?

Maybe I couls sold out the LED on Pin 2.

Thank you!

peterhinch, 2019-07-04:
It does sound as if your DS3231 has pullups; as long as there are pullups somewhere you'll be OK. I have used pins 0 and 2 for I2C communication between a Pyboard and an ESP8266 [here](https://github.com/peterhinch/micropython-async/tree/master/i2c). I chose those pins so that users would not normally need to worry about pullups and because I knew that the I2C link to the Pyboard would not affect the boot process.

As far as I'm aware you can use any pin, but you'll need advice from others about re-purposing the UART pins. You'll need to make sure that, when you boot, pins 0 and 2 are not pulled down by any of your kit.