peterhinch/micropython-samples #17: Make micropip.py output a helpful message when installing a CPython package is attempted

aivarannamaa, 2020-09-12:
I'm not sure whether this work should start from `upip` or here, but I'll raise the issue anyway.

At the moment micropip.py (and probably also upip) doesn't give a meaningful error message when user tries to install a non-micropython package (for an example see https://github.com/thonny/thonny/issues/1330).

It would be nice if micropip.py anticipated it and said something like "Package X doesn't look like a MicroPython package"

peterhinch, 2020-09-12:
Do you have any ideas on how this might be achieved?

> I'm not sure whether this work should start from upip or here

To answer this **micropip** is a port of **upip**: if its maintainers implement new features I will aim to migrate them to **micropip**. It might therefore be best to raise the issue against **upip**. Further, its developers may know a way to actually accomplish this.

aivarannamaa, 2020-09-20:
> Do you have any ideas on how this might be achieved?

What about catching all errors when the package name doesn't start with a known good prefix ("micropython-", "pycopy-") and appending "Are you sure this package is meant for MicroPython?" to the error message?

peterhinch, 2020-09-20:
Possibly. As I said I don't want to make `micropip` functionality differ from that of `upip`. I therefore suggest you raise enhancement requests against `upip`. If they are accepted, I will port them.