peterhinch/micropython-samples #19: Calibration value out of range

Gbouna, 2021-09-06:
Thanks for the library, its helpful to get my RTC working. 

However, when trying to calibrate the RTC, it gave the error saying calibration value is out of range. Any suggestions on how to resolve this?

peterhinch, 2021-09-06:
What hardware are you using?

Gbouna, 2021-09-07:
Thanks for the prompt response, I am using OpenMV Cam H7

Gbouna, 2021-09-07:
And the RTC i am using is DS3231

peterhinch, 2021-09-07:
OpenMV Cam H7 uses different hardware and different firmware. It's hard for me to comment as I have no knowledge of the H7 chips. It would need a detailed study of the chip datasheet to figure out whether calibration works in the same way as the chips used on the various Pyboards - your observation suggests that it does not.

I can't support hardware I don't possess. Calibration is specific to Pyboards. 

You should be able to use my code to read the DS3231. It may be that the best you can do is to periodically synchronise the chip's RTC to the DS3231.

closed 2021-09-07 by peterhinch

Gbouna, 2021-09-07:
Thanks for your response and suggestions

Actually what is important for me that I want to do is to be able to set the RTC time according to the current time and save it to the back up memory so that when there is power outage, the time will still be accurate since the RTC has a back up battery.

Any idea on how to do this will be appreciated.

Thanks

peterhinch, 2021-09-07:
I'm not sure I follow. After an outage, read the DS3231 (which has a battery and so maintains time) and use it to set the host's RTC.

Gbouna, 2021-09-07:
The reason I want to use the external RTC in my application is for accurate keeping of time, and my application requires that the OpenMV Cam goes to sleep at certain time and then wake up. Usually with the onboard RTC, when the board wakes up, the time and date reset itself back to how i set it originally. I am hoping to use external RTC so that the time and date is preserved in the backup memory of the RTC such when the board wakes up, the time will still be accurate.

Gbouna, 2021-09-07:
Your library is still very useful for me, if I can just know how to set the DS3231 RTC time to the current time and save it, such that I can just read from it, and it will keep running without the need to set again even after power outage of the OpenMV Cam, then it will be cool for me. I am thinking that this is possible since the DS3231 RTC has its own backup battery.

peterhinch, 2021-09-07:
It's hard for me to comment as I don't have your hardware. I would start out by following the docs for a Pyboard. Set the board's RTC and run `save_time()` to set the DS3231. After an outage use `get_time(True)` to set the board's RTC from the DS3231. If this doesn't work you may need to seek advice from people with your hardware.