peterhinch/micropython-samples #20: ds3231 calibration for Pyboard 1.x seems broken.

iabdalkader, 2021-09-12:
https://github.com/peterhinch/micropython-samples/blob/0d27d95e8084acc657944d0b4aba59758d447977/DS3231/ds3231_pb.py#L113

The calibration routine for 1.x boards seems to be broken. I made a minor change to ds3231_pb.py to force using get_cal (1.x boards):
```Python
d_series = os.uname().machine.split(' ')[0][:4] == 'PYBD'
if d_series:
    machine.Pin.board.EN_3V3.value(1)
d_series = False # Force 1.x get_cal
````

And running the following script using a PYBD_SF2 results in out of range calibration value:

```Python
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== from ds3231_pb import DS3231
=== from machine import I2C
=== 
=== ds3231 = DS3231(I2C(2))
=== print("DS3231 RTC Time:", ds3231.get_time(set_rtc=True))
=== 
=== import time
=== print(time.localtime())
=== 
=== ds3231.calibrate()
=== 
DS3231 RTC Time: (2021, 9, 12, 7, 42, 48, 0, 0)
(2021, 9, 12, 7, 42, 48, 0, 255)
Pyboard 1.x. Waiting 5 minutes for calibration factor.
Error -6332.2ppm -12050.1mins/year. Cal factor 6637
Traceback (most recent call last):
  File "<stdin>", line 10, in <module>
  File "ds3231_pb.py", line 182, in calibrate
ValueError: calibration value out of range
```

I think I have one of the very first boards could dig it up and test with if needed.

peterhinch, 2021-09-13:
I don't understand why you changed it. This has been tested against Pyboard D SF2W and SF6W, as described in the doc.

iabdalkader, 2021-09-13:
> Why do you want to use Pyboard 1.1 code against a Pyboard D?

I don't, I just wanted to test the 1.x function and all I had was a D board (SF2W).

peterhinch, 2021-09-13:
I can assure you that the calibration routine does work when run against a Pyboard 1.x.

closed 2021-09-13 by peterhinch

iabdalkader, 2021-09-13:
> I can assure you that the calibration routine does work when run against a Pyboard 1.x.

Evidently it's not:

```Python
MicroPython v1.17-20-g0a5107372-dirty on 2021-09-13; PYBLITEv1.0 with STM32F411RE
Type "help()" for more information.
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== from ds3231_pb import DS3231
=== from machine import I2C
=== 
=== ds3231 = DS3231(I2C(3))
=== print("DS3231 RTC Time:", ds3231.get_time(set_rtc=True))
=== 
=== import time
=== print(time.localtime())
=== 
=== ds3231.calibrate()
=== 
DS3231 RTC Time: (2021, 9, 13, 5, 36, 39, 1, 0)
(2021, 9, 13, 5, 36, 39, 1, 256)
Pyboard 1.x. Waiting 5 minutes for calibration factor.
Error -52822.7ppm -100521.5mins/year. Cal factor 55369
Traceback (most recent call last):
  File "<stdin>", line 10, in <module>
  File "ds3231_pb.py", line 183, in calibrate
ValueError: calibration value out of range
```

[image]

I can help with testing if you are interested in fixing it, otherwise just ignore my comment.

peterhinch, 2021-09-13:
The Pyboard Lite does not have a crystal controlled clock so cannot be calibrated. It works when run against a Pyboard 1.0 or 1.1, which do have crystals and can run to timepiece levels of precision when calibrated.

iabdalkader, 2021-09-13:
> The Pyboard Lite does not have a crystal controlled clock so cannot be calibrated

Oh I see the issue now,  there's a footprint for the 32Khz crystal maybe I'll try it later, thanks for your time and sorry for the noise :neutral_face: