peterhinch/micropython-samples #21: Broken url/link on micropython-samples (27.09.2021)

beyonlo, 2021-09-27:
Hi Peter!

In the link https://github.com/peterhinch/micropython-samples/blob/master/SERIALISATION.md has a "The extremely simple source may be found in the official library. " where "the official library" is a hyperlink to https://github.com/micropython/micropython-lib/tree/master/pickle where this link do not exists.

Thank you.

peterhinch, 2021-09-28:
Thanks for the report - you are being very thorough!

The official library has been reorganised. I've updated my link.

closed 2021-09-28 by peterhinch

beyonlo, 2021-09-28:
@peterhinch I like so much your work (uasyncio docs and examples, like as everything other samples). I'm reading and learning a lot with your examples and docs. So, always that I'm reading and find a broken link, I opening a issue - this is a way to me to do a little contribution for your work.

Thank you.