peterhinch/micropython-samples #27: all packages throw 403 error

robvoi, 2022-01-28:
I can't install any micropython package. All packages throw an error 403. The json in the pypi URL exists.
I have no idea on where to start searching. Can someone maybe point me to the right direction? Thanks!

python3 -m micropip install micropython-rosserial

results in:

Installing to: /home/robo/Downloads/
Queue: ['micropython-rosserial']
https://micropython.org/pi/micropython-rosserial/json
Warning: micropython.org SSL certificate is not validated
https://pypi.org/pypi/micropython-rosserial/json
Error installing 'micropython-rosserial': b'403', packages may be partially installed

peterhinch, 2022-01-29:
I'm not sure either. My test script works:
```bash
$ micropip.py install -p ~/rats micropython-copy
Installing to: /home/adminpete/rats/
Warning: micropython.org SSL certificate is not validated
Installing micropython-copy 3.3.3.post2 from https://micropython.org/pi/copy/copy-3.3.3.post2.tar.gz
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython-samples/micropip
$ cd
[adminpete@capybara]: ~
$ cd rats/
[adminpete@capybara]: ~/rats
$ ll
total 12
-rw-rw-r-- 1 adminpete adminpete 8859 Jan 29 10:19 copy.py
```
The `micropip.py` script is a port of the official `upip.py` so I attempted to install your package to an ESP8266 using `upip`. This failed:
```python
>>> upip.install("micropython-rosserial")
Installing to: /lib/
Warning: micropython.org SSL certificate is not validated
Installing micropython-rosserial 0.2.8 from https://files.pythonhosted.org/packages/c5/e0/3b232b334011948b543811d3b9ed2e9ad9973ec2aa27dcb6b42156a082f3/micropython-rosserial-0.2.8.tar.gz
Error installing 'micropython-rosserial': [Errno 22] EINVAL, packages may be partially installed
>>> 
```
I then went on to install "mcropython-copy" and to check the outcome of both installs.
```python
>>> upip.install("micropython-copy")
Installing to: /lib/
Installing micropython-copy 3.3.3.post2 from https://micropython.org/pi/copy/copy-3.3.3.post2.tar.gz
>>>  
/home/adminpete/Downloads> ls /pyboard/lib/
copy.py
/home/adminpete/Downloads> 
```
So it correctly installed `copy.py` but did nothing with `micropython-rosserial`.

This suggests that either there is a problem with the package, or there is a bug or limitation in `upip`. I am not knowledgeable about pypi: if you are convinced that the package is valid, it might be best to raise an issue against `upip` and I will port any changes the maintainers make to `micropip`.

Sorry I can't offer anything more authoritative.

robvoi, 2022-01-29:
Thanks for the swift reply and your time for testing it.
I found your script as it woks for the pi pico which doesn't have wifi. That's why I got stuck with upip and hoped to have a solution with your script.

peterhinch, 2022-01-29:
It is possible to run official `upip` under the Unix build of MicroPython, but on the basis of my test this won't fix your problem. You might ask in [the forum](http://forum.micropython.org/index.php): maybe someone will have an idea of why `upip` can't access that repository.

However, as a straight port of `upip`, I will only implement changes to `micropip` in response to improvements in `upip`.

closed 2022-01-29 by peterhinch

peterhinch, 2022-01-31:
I have updated the micropip README to explain this issue.