peterhinch/micropython-samples #35: DS3231 wrong strip of Century

heikokue, 2023-05-02:
lines 78-81 of ds3231_gen.py are:
`self.ds3231.readfrom_mem_into(_ADDR, 0, data)`
`ss, mm, hh, wday, DD, MM, YY = [bcd2dec(x) for x in data]`
`MM &= 0x1F  # Strip century`
`YY += 2000`
For a correct strip of the century bit 7 of the MM register, it needs to be done before bcd2dec and not after.

peterhinch, 2023-05-02:
The century bit has already been stripped in `bcd2dec()` here `(bcd & 0x70) >> 4`. So the line
```py
MM &= 0x1F
```
does nothing and can be deleted.

closed 2023-05-02 by peterhinch

heikokue, 2023-05-02:
Hello Peter,
I'm sorry, but I found another bug.
Here the code:
`        def bcd2dec(bcd):  # Strip MSB`
`            return ((bcd & 0x70) >> 4) * 10 + (bcd & 0x0F)`
`        self.ds3231.readfrom_mem_into(_ADDR, 0, data)`
`        ss, mm, hh, wday, DD, MM, YY = [bcd2dec(x) for x in data]`
The YY register uses bit 7.
So it's no good idea to strip it in bcd2dec().
The bug is not very critical, because it appears first 2080 :relaxed:

peterhinch, 2023-05-02:
I was aware of this. It's easily fixed, and I'll do it as a 130th birthday present to myself. I like to have something to look forward to :)