peterhinch/micropython-samples #40: Code not working after powering off Raspberry Pico

pulcesplosiva, 2024-09-14:
The sample code encoders/encoder_rp2.py works great until you power off and on the Pico. Even if you try to reflash the code it does not work, and no exceptions are being detected.

The only way I found to make it work again, is to flash another program (just some random code I had at hand) and than reload this sample code.

Than again...if I power off the pico the same issue returns. Seems like I'm the only one experiencing this problem, but I'm not understanding if it is a micropython issue or a hardware issue.

pulcesplosiva, 2024-09-16:
I managed to solve this problem thanks to a user on reddit. For some reason the sample code works great the first time after loading it into the pico...after that on the next power cycle it stops detecting the encoder pulses.

The reason for this was because the pins where not set in pull up mode. 

Still a mystery to me why the pins are pulled up after flashing the pico tho.

peterhinch, 2024-09-16:
Good point, thanks for posting that. I've added a code comment to that effect.

closed 2024-09-16 by peterhinch