peterhinch/micropython-samples #41: DST offset being set incorrectly in date.py?

pointy56, 2024-11-04:
Hi Peter,
I have been looking for a 'good' solution for DST handling for an embedded controller based on a Pico W.
I am using NTP to maintain the RTC as UTC, and am applying DST correction when displaying or using the current time.

I came across a posting in the Micropython forum that mentioned date.py, and liked it's simplicity as I only need to support BST.
I have been testing it out, and found a case where I believe time_offset returns the wrong value:

My understanding is that BST starts at 01:00 on the last Sunday of March and ends at 02:00 on the last Sunday in October.
If, using your example code, I invoke time_offset with on 31/03/2024 with hour_utc set to 1 it returns a value of 0.
I believe that it should return 1 (if I try with hour_utc set to 1.000001 it returns 1).  Other cases appear to return the correct values.

I tried tweaking the code so that the various edge cases return the correct value, and ended up changing the code as follows:

        if m < 3:
            return m  # Deduce time offset from month alone
        hr_chg = (hr < 1) if m == 3 else (hr <= 1)
        return int(
            ((self._lt[2] < mday_sun) or (self._lt[2] == mday_sun and hr_chg)) ^ (m == 3)
        )  # Months where offset changes

I believe that my interpretation of the DST offset is correct, hr_chg allows for handling 01:00 differently at BST start & end
You may well be able to come up with a more elegant solution, but this seems to work for me.

I will be interested in your feedback.
Martin

peterhinch, 2024-11-05:
I am unconvinced. See [timeanddate.com](https://www.timeanddate.com/time/change/uk). This indicates that DST changes at 01:00UTC. When the clocks go back in Autumn, this occurs at 02:00 _on the clocks_, i.e. 01:00UTC. This is as per my docs:

> If an hour value is passed, the change will occur at the correct 01:00 UTC.

On a microcontroller it is usual to run the hardware clock on UTC, adjusting the time for display or alarms to local time as required (it is bad practice to have sudden changes in machine time). That is the underlying reason my DST routine takes UTC as input. Your version is correct for a system whose clock is adjusted to local time.

closed 2024-11-05 by peterhinch

peterhinch, 2024-11-07:
Please run the following script:
```py
from date import DateCal
d = DateCal()
d.now()
d.month = 10
d.mday = 27
for m in range(0, 120, 5):
    t = m/60  # Hours as a float
    h = m // 60  # Integer hours
    im = m % 60  # Minutes for display
    print(f"{h:02d}:{im:02d} {t:3.1f} Offset: {d.time_offset(t)}")
```
This tests the most recent DST change. It shows that the change occurs at 01:00 UTC. You can zoom in closer to the transition with
```py
from date import DateCal
d = DateCal()
d.now()
d.month = 10
d.mday = 27
for m in range(55, 65):
    t = m/60  # Hours as a float
    h = m // 60  # Integer hours
    im = m % 60  # Minutes for display
    print(f"{h:02d}:{im:02d} {t:3.1f} Offset: {d.time_offset(t)}")
```
with the following outcome:
```
00:55 0.9 Offset: 1
00:56 0.9 Offset: 1
00:57 1.0 Offset: 1
00:58 1.0 Offset: 1
00:59 1.0 Offset: 1
01:00 1.0 Offset: 1
01:01 1.0 Offset: 0
01:02 1.0 Offset: 0
01:03 1.1 Offset: 0
01:04 1.1 Offset: 0
```

pointy56, 2024-11-07:
I hear what you're saying, but the way it's described on the UK government website is: In the UK the clocks go forward 1 hour **at** 1am on the last Sunday in March, and back 1 hour **at** 2am on the last Sunday in October (bold is mine).  Similar wording can be seen on timeanddate.com: When local standard time was about to reach Sunday, 31 March 2024, 01:00:00 clocks were turned forward 1 hour to Sunday, 31 March 2024, 02:00:00 local daylight time instead **and** When local daylight time was about to reach Sunday, 27 October 2024, 02:00:00 clocks were turned backward 1 hour to Sunday, 27 October 2024, 01:00:00 local standard time instead.

This is how I've always understood and experienced it on numerous different systems over my career in IT.
Anyway, I am very happy to use your code, it's really useful, but I think that I will stick with hr<1 - we can agree to differ!

On a separate point, I have been using your code with full Python as my laptop is always on.  I realised that I needed to remove the const keyword, but it barfed at self._cur = mktime(self._lt) as it was expecting a tuple - easily fixed, but I was surprised that Micropython didn't require tuple(self._lt) in the same way; do you think is this expected behaviour or a bug?

peterhinch, 2024-11-07:
The above description is correct from the standpoint of someone adjusting a clock which tells local time. My definition is correct from the standpoint of someone with a clock that continuously runs UTC. The user wishes to display local time without actually changing the clock. My POV reflects the intended application.

I have updated the doc in an attempt to clarify this and to explain the reasoning. I have also pushed a code change to enable the module to run under CPython.

I'm not sure why CPython's `mktime` barfs on a `list`. It strikes me as unreasonably picky :)

pointy56, 2024-11-07:
Sounds good - thank you.