peterhinch/micropython-samples #42: current sun height in degrees

sgbmzm, 2024-12-12:
Hello.
Is it possible that you could add a calculation of the current sun height in degrees?
Thank you!

peterhinch, 2024-12-12:
The method `RiSet.sin_alt()` returns the sine of the altitude of the sun or moon - see [code](https://github.com/peterhinch/micropython-samples/blob/d2929df1b4556e71fcfd7d83afd9cf3ffd98fdac/astronomy/sun_moon.py#L355). It takes the following args:
* `hour` Hour of day.
* `sun` Set `True` for Sun.

Latitude and longitude and current date need to be set as per the docs. You should be able to extract the angle in degrees at any time in the current day with
```py
math.asin(riset_instance.sin_alt(hour, True)) * 180 / math.pi
```

sgbmzm, 2024-12-13:
Thank you very much! But I can't seem to correctly define the "now" time to get a correct result.
Could you please give an example where the "now" time would be defined?
Many thanks
```
import sun_moon
location = {"name": "modiin-illit", "lat": 31.94, "long": 35.03, "utc_offset": 2}
riset_instance = sun_moon.RiSet(lat=location["lat"], long=location["long"], lto=location["utc_offset"])
# now = ?????? 
sun_alt_degree = math.asin(riset_instance.sin_alt(now, True)) * 180 / math.pi
```

peterhinch, 2024-12-13:
I ran the following script. Note that defaults (lat, long, lto) happen to be correct for my location and machine datetime is correct.
```py
import math
from sun_moon import RiSet
rs = RiSet()
print(f" sunrise: {rs.sunrise(2)} sunset {rs.sunset(2)}")

def altitude(hour):
    return math.asin(rs.sin_alt(hour, True)) * 180 / math.pi

for hour in range(7, 17):
    print(f"Hour {hour} altitude {altitude(hour)}")
```
The outcome was
```py
Machine time: 13/12/2024 09:12:26
 sunrise: 08:15:31 sunset 15:49:58
Hour 7 altitude -10.35806861208352
Hour 8 altitude -2.678344028278478
Hour 9 altitude 3.90032128901863
Hour 10 altitude 9.011564346937938
Hour 11 altitude 12.30255254252035
Hour 12 altitude 13.50301296029709
Hour 13 altitude 12.50412447500261
Hour 14 altitude 9.396757722209848
Hour 15 altitude 4.439255212092692
Hour 16 altitude -2.019940920301286
```

sgbmzm, 2024-12-13:
Thanks! It works for me but the data is correct For the requested hour only when I subtract UTC_OFFSET
I also learned that you can use math.degrees() directly
Here is the code that worked for me
```
import math
from sun_moon import RiSet
location = {"name": "modiin-illit", "lat": 31.94, "long": 35.03, "utc_offset": 2}
rs = RiSet(lat=location["lat"], long=location["long"], lto=location["utc_offset"])
print(f" sunrise: {rs.sunrise(2)} sunset {rs.sunset(2)}")

def altitude(hour):
    #return math.asin(rs.sin_alt(hour, True)) * 180 / math.pi
    return math.degrees(math.asin(rs.sin_alt(hour, True)))

for hour in range(7, 17):
    print(f'Hour {hour} altitude {altitude(hour-location["utc_offset"])}')

```

peterhinch, 2024-12-14:
I'm puzzled by that. I agree that `math.degrees` is a neater solution (and is correct), but it produces the same result as my approach:
```py
>>> import math
>>> theta = math.pi/4
>>> math.degrees(theta)
45.0
>>> theta * 180 / math.pi
45.0
>>> 
```

sgbmzm, 2024-12-14:
True. It's the same thing as you said, only it's a neater solution (it didn't come from me, but from chat gpt).

In any case, I really enjoy your valuable library, and thank you very much for that, so I would very much like a few more things.
1. Can you add a function to get the azimuth (in degrees from the north) of the sun/moon for a given hour. Or maybe add this into the `sin_alt()` function to be `sin_alt_az()`
2. Add an optional variable into the above function such as `sin_alt(self, hour, sun, degrees=False)` that will allow you to directly get the result in degrees of both the altitude and the azimuth.
3. Add an option to the sunrise and sunset output such as `variants: 3`  that will return a tuple of date and time in `datetime` format

peterhinch, 2024-12-15:
Taking the last point first you can get a datetime tuple with
```py
time.localtime(rs.sunrise(1))
```
Re azimuth, this can readily be calculated.

Consider the northern hemisphere.

On the Greenwich meridian at 12:00:00 UTC the sun is due South. Observers on the same longitude will also see the sun at due South (with differing altitudes). Azimuth is independent of longitude. Observers east or west of Greenwich will see the sun at due south at around noon local time. Azimuth at 12:00:00 UTC depends on longitude (and was famously a way of measuring it). For every degree you move west the apparent position of the sun moves east by one degree. (Concorde passengers flying from London to New York would see the sun appear to set in the East only to rise again at a later time (in the East)).

Azimuth (at Greenwich and elsewhere) rotates clockwise by 360° every 24 hours as the earth rotates, i.e. 15° per hour.

Things get interesting at the equator where at 12:00:00 UTC the sun is famously overhead; so doesn't really have an azimuth. If you continue south, the noon azimuth is North with the apparent direction of the sun's motion being anticlockwise. If clocks had been invented in Australia they would have run anticlockwise.

For the northern hemisphere you want something like
```py
def azimuth(h):  # h is hour of day, 0..23.99 UTC
    az = (h - 12) * 15  # -ve is east, 0 is South
    return az + longitude_correction  # one degree per degree longitude
```

As a general point, items I publish in "samples" are intended as ideas for development and experimentation and are not necessarily "complete". As per the first paragraph of the docs "Egregious bugs will be fixed but I may not accept feature requests."

sgbmzm, 2024-12-15:
I think this method will not help to know the azimuth of the moon but only the sun.
If in the future you have time for these additions to the main code, I would be happy.
Anyway, thank you very much!.

sgbmzm, 2025-02-16:
The current moon altitude calculation is much less accurate than the sun altitude. Do you have any idea why this is happening? And how to fix it?

sgbmzm, 2025-02-24:
See some of the solutions in https://github.com/peterhinch/micropython-samples/issues/50
But I would still love to know why The Current Moon Altitude Calculation is Much Less Accoraate Thn The Sun Altitude.
Is it because of the inaccuracy of the `minimoon` function?