peterhinch/micropython-samples #44: Encoder documentation does not explain how encoder_rp2.py works with one input pin

sk8board, 2024-12-20:
Hello Peter,

Thank you for sharing your excellent work and many years of experience.

I have read through your [encoder](https://github.com/peterhinch/micropython-samples/blob/master/encoders/ENCODERS.md) documentation. While this was informative, the documentation naturally assumes two inputs of a quadrature encoder. 

Your `encoder_rp2.py` module only uses one input of a quadrature encoder, but your documentation does not explain how and why one input is used.

Could you explain:

1) why one input rather than two inputs of a quadrature encoder are used in `encoder_rp2.py`?

2) how are you getting away with using only one input?

3) could you explain v, x, y, s variables in the ISR function?

4) could you explain what the math calculations in the ISR function are doing?

sk8board, 2024-12-20:
``` py
# encoder_rp2.py Uses the PIO for rapid response on RP2 chips (Pico)

# Copyright (c) 2022 Peter Hinch
# Released under the MIT License (MIT) - see LICENSE file

# PIO and SM code written by Sandor Attila Gerendi (@sanyi)
# https://github.com/micropython/micropython/pull/6894

from machine import Pin
from array import array
import rp2

# Test with encoder on pins 2 and 3:
# e = Encoder(0, Pin(2))

# while True:
# time.sleep(1)
# print(e.value())

# Closure enables Viper to retain state. Currently (V1.17) nonlocal doesn't
# work: https://github.com/micropython/micropython/issues/8086
# so using arrays.
def make_isr(pos):
    old_x = array("i", (0,))

    @micropython.viper
    def isr(sm):
        i = ptr32(pos)
        p = ptr32(old_x)
        while sm.rx_fifo():
            v: int = int(sm.get()) & 3
            x: int = v & 1
            y: int = v >> 1
            s: int = 1 if (x ^ y) else -1
            i[0] = i[0] + (s if (x ^ p[0]) else (0 - s))
            p[0] = x

    return isr

# Args:
# StateMachine no. (0-7): each instance must have a different sm_no.
# An initialised input Pin: this and the next pin are the encoder interface.
# Pins must have pullups (internal or, preferably, low value 1KΩ to 3.3V).
class Encoder:
    def __init__(self, sm_no, base_pin, scale=1):
        self.scale = scale
        self._pos = array("i", (0,))  # [pos]
        self.sm = rp2.StateMachine(sm_no, self.pio_quadrature, in_base=base_pin)
        self.sm.irq(make_isr(self._pos))  # Instantiate the closure
        self.sm.exec("set(y, 99)")  # Initialise y: guarantee different to the input
        self.sm.active(1)

    @rp2.asm_pio()
    def pio_quadrature(in_init=rp2.PIO.IN_LOW):
        wrap_target()
        label("again")
        in_(pins, 2)
        mov(x, isr)
        jmp(x_not_y, "push_data")
        mov(isr, null)
        jmp("again")
        label("push_data")
        push()
        irq(block, rel(0))
        mov(y, x)
        wrap()

    def position(self, value=None):
        if value is not None:
            self._pos[0] = round(value / self.scale)
        return self._pos[0] * self.scale

    def value(self, value=None):
        if value is not None:
            self._pos[0] = value
        return self._pos[0]
```

sk8board, 2024-12-20:
``` py
v: int = sm.get() & 3				# bit mask any 1's greater than 3, 0000011
						# result is 0,1,2, or 3

x: int = v & 1					# if 3, then x = 1,
						# if 2, then x = 0,
						# if 1, then x = 1,
						# if 1, then x = 0,
						# if odd, then 1,
						# if even, then 0
						# result is X input value

y: int = v >> 1					# if 3, then y = 1,
						# if 2, then y = 1,
						# if 1, then y = 0,
						# if 0, then y = 0,
						# if 2 or 3, then 1,
						# if 1 or 0, then 0
						# result is Y input value

s: int = 1 if (x ^ y) else -1			# if x = 0 and y = 1, then s = -1, else s = 1,
						# three states result with 1,
						# one state results with -1

i[0] = i[0] + (s if (x ^ p[0]) else (0 - s))	# if x = 0 and p[0] = 1, then -s else s
						# add or subtract 1 to i[0] (position)

p[0] = x					# save x to old_x
```

### Question: 
Where is `old_y` within the ISR function? According to your [truth table](https://github.com/peterhinch/micropython-samples/blob/master/encoders/ENCODERS.md#7-algorithm) four values are needed, `x`, `y`, `old_x`, and `old_y`.

peterhinch, 2024-12-21:
The file `encoder_rp2.py` uses the PIO and therefore works differently from my interrupt based versions. The state diagram refers to the interrupt based solution. In the code comments you'll see
```py
# Test with encoder on pins 2 and 3:
# e = Encoder(0, Pin(2))
```
The PIO has the concept of a "base pin" with any other pins being subsequently numbered.

I suggest you read up on PIO programming. There is a lot of good documentation on the Raspberry Pi site.

The PIO version has the potential to cope with high speed signals. Alternatively the interrupt driven solution works fine on the Pico.

sk8board, 2024-12-22:
> The file `encoder_rp2.py` uses the PIO and therefore works differently from my interrupt based versions. The state diagram refers to the interrupt based solution. In the code comments you'll see

Thank you. I made this graphic in a spreadsheet which helped show why `old_y` is not needed.

[image]

> The PIO has the concept of a "base pin" with any other pins being subsequently numbered.
>
> I suggest you read up on PIO programming. There is a lot of good documentation on the Raspberry Pi site. [...]

Yes, this and DMA is why I prefer the Pico over ESP32.

Thank you for sharing your many years of experience!!!

closed 2024-12-22 by sk8board