peterhinch/micropython-samples #45: Functor example

32BT, 2025-01-29:
Hi Peter,

Great knowledgebase. 

I recently ventured on controlling the RP2040 by its registers (via Python, not via the C-sdk), just to get better acquainted with its internals. I chose to control the ADC as a representative example. The RP2040 has a single ADC unit, which can be reset as a unit, and then it can be enabled to sample one of five pads. Seemed like a reasonable usecase for a singleton-like construct, not least because the controller object can then guard against simultaneous requests from multiple threads for example.

From the user's perspective I wanted to implement the following pattern:

```Python
import hw

# Use classmethod to reset hw unit, since this is a property of the hw, not of the ADC (instance)
hw.ADC.reset_block() 

# Use (lazy) singleton to occasionally take a sample and power everything down again
with hw.ADC(enable_temp=True) as adc:
    print(adc.sample_vsys())
    print(adc.sample_temp())
```

I eventually ended up writing a functor  which brings me to a potential pitfall I saw in the examplecode.
I believe that in the example the following line would return "None" instead of the instance. 
https://github.com/peterhinch/micropython-samples/blob/d2929df1b4556e71fcfd7d83afd9cf3ffd98fdac/functor_singleton/README.md?plain=1#L70

In your example code that is not a problem, it might even be intentional, but for a contextmanager construct that would probably not fly well. I just wanted to mention that for the sake of completeness.  

In my own code I also wanted the ADC object to be as simple as possible, so I let the functor add the "__call__" statement which then wires back to "__init__" under the assumption that init should merely prepare an object for running, not actually start it running. I'm now brewing on the following line:
https://github.com/peterhinch/micropython-samples/blob/d2929df1b4556e71fcfd7d83afd9cf3ffd98fdac/functor_singleton/README.md?plain=1#L109

This kind of feels like (if you can forgive me a metaphor): I go get my motorcycle out of my shed, at which point it kickstarts itself and starts driving just because I brushed off the dust. I'm sure there is an electric car manufacturer somewhere who thinks that is exactly how it should work, but wat can I say: I am not an electric car manufacturer...

Anyway, thanks for a great resource.

peterhinch, 2025-01-30:
I hadn't considered the case of combining functors and context managers. I'm still unclear on the circumstances in which line 70 can return `None`. Please could you post an example exhibiting the behaviour based on my code sample.

32BT, 2025-01-30:
```Python
@functor
class MyFunctor:
    def __init__(self, arg):
        self.state = arg
        print('In __init__', arg)

    def __call__(self, arg):
        print('In __call__', arg + self.state)

    def __enter__(self):
        return self

    def __exit__(self, *args):
        pass

MyFunctor(42)  # prints 'In __init__ 42'
MyFunctor(5)  # 'In __call__ 47'

with MyFunctor(4) as f:
    pass

```

which leads to the following output: 
```
In __init__ 42
In __call__ 47
In __call__ 46
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "test.py", line 29, in <module>
AttributeError: 'NoneType' object has no attribute '__exit__'
```

because ```__call__``` does not return anything specific thereby defaulting to returning None.

peterhinch, 2025-01-31:
Thank you for clarifying. You observation is correct.

I have pushed an update. With hindsight it seems quite clear that `--call__` should return `self`!

closed 2025-01-31 by peterhinch