peterhinch/micropython-samples #46: DeepSeek was hallucinating about you. I wish it would be right somehow

MichaelMuckKremtz, 2025-01-30:
Have you ever thought about kinetic installations with the Pi Picos PIO as superfast Step Generators and UART configuration of the TMC2209 chip?

I hope that reasoning models soon will be capable of implementing stuff like that. 

[image]

peterhinch, 2025-01-31:
How deeply bizarre. As you are evidently aware, I have done nothing with stepper motors on MP or elsewhere. In fact the last time I recall using one was in 1973.

LLM answers to technical queries tend to fall into the categories "utterly wrong" or "laughably generic". The above achieves both.

MichaelMuckKremtz, 2025-02-01:
In this case your right but I think it's worth testing new LLMs from time to time since they are getting better.

I personally did some small projects that were above my poor programming skills but I could manage to make them work with the help of those systems and learned along the way.

peterhinch, 2025-02-02:
I do. I've tested them with a queries about realtime programming - as might appear on a university CS course. The response of LLM's to that query has definitely improved. My score has gone from one out of ten to about eight.

Which seems pretty good. Alas each time I've had an actual technical problem where I needed an answer, I've learnt nothing.

In an article I read today a problem I've been predicting for some time has come to pass. The training data for the latest generation of LLM's includes the absolute shite excreted by the earlier generations with predictable results 🥇

MichaelMuckKremtz, 2025-02-03:
Since two days I was experimenting quite a lot with o3-mini-high and achieved excellent results with micropython and my TMC2209 stepper driver use case. Really happy. Now I have the licence for one month and will try to make as much progress as possible. :-)

I'll probably throw some of your great documentation on the model as well. Hope your are fine with that.

All the best from Cologne!