peterhinch/micropython-samples #47: NTPTIME [Errno 116] ETIMEDOUT OR 0

sgbmzm, 2025-02-10:
Hello.
In my T-DISPLAY-S3 I am trying to get the time from NTP in both the regular directory and your directory.
It only rarely works. Most of the time I get an **[Errno 116] ETIMEDOUT** error when I use the regular directory or the time is **0** i.e. EPOCH when I use your directory. How do I solve this problem?
Thank you!

peterhinch, 2025-02-11:
NTP clients expect the time server to respond quickly - there is no point in an outdated time sample. I would suspect network or internet latency.

A solution is to run a local timeserver. This is very easy: I have an ancient Raspberry Pi running the `ntp` daemon which provides low-latency lookups.

sgbmzm, 2025-02-16:
Thanks, but local timeserver won't help me.